wfdudley/T-watch-2020 #6: Sometimes the hour hand in the analog clock points wrong.

wfdudley, 2020-10-09:
This seems to self correct after a few minutes.  Seems most likely to happen just before the hour change.

Therealjosephchrzempiec, 2021-01-06:
I get the same problem. WIth the original code from Dan Geiger. When i try to set the time say for 03.00- 3am i get 13:00 or 1pm. I haven't tried wfududly yet. 

Joseph

wfdudley, 2021-01-06:
Not the same problem.

The bug I reported is that the hour hand will briefly point wrong, but it
corrects itself, probably when the next minute starts.

Bill Dudley

This email is free of malware because I run Linux.

Therealjosephchrzempiec, 2021-01-06:
I will have to check that out. I guess because i didn't check it before the minute was up. I just saw it and freaked out.

Joseph