wfdudley/T-watch-2020 #7: BLE Beacon

anarchking, 2020-11-01:
I  think it would be very useful to have an app that you click a button and it turns on the ble Beacon Broadcasting so other devices can be scanning in all rooms so you're so you're smart home can recognize what room you're in.

Example: https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_iBeacon/BLE_iBeacon.ino

wfdudley, 2020-11-01:
I've never heard of that.  I'll research that.  My biggest worry is that it
will run the battery down quickly.

Bill Dudley

This email is free of malware because I run Linux.

anarchking, 2020-11-01:
I was thinking a way to make a option to turn it on when connected to home network or mqtt server and only run when motion is active, maybe even a timer for not running in the middle of the night.

wfdudley, 2020-11-01:
One problem is that the watch goes to sleep regularly, or the battery will
quickly die.
I don't know how to keep the bluetooth system alive during those sleep
times.
I don't see how this can work.  But I'm also not expert enough in power
management on
the T-watch 2020 to be able to speak authoritatively.

Bill Dudley

This email is free of malware because I run Linux.

wfdudley, 2020-11-13:
Since I don't know anything about the problem space, and have no way to test if it works or not, I won't be working on this.

closed 2020-11-13 by wfdudley