wfdudley/T-watch-2020 #8: Seems broken to me?

lachesis, 2020-11-03:
Hi, maybe I've gotten something very wrong here, but this project doesn't seem to work to me. I was able to build and flash using the Arduino IDE after changing the name of the folder to "DudleyWatch" and installing the following libraries:
```
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library
https://github.com/bxparks/AceTime
https://github.com/Bodmer/TFT_eSPI
https://github.com/bxparks/AceCommon
https://github.com/bblanchon/ArduinoJson
https://github.com/arduino-libraries/Ethernet
https://github.com/knolleary/pubsubclient/
https://github.com/earlephilhower/ESP8266Audio
```

That's where my problems started. None of the given gestures seem to work for me. I seem to have access to only two things:
(1) the MQTT app (which is proceeded by a "scan networks" page which does seem able to connect to my wifi), accessed by swiping in any direction on the home screen
(2) the settings app, accessed by CCW rotation

I can't seem to launch the app select screen reliably - most of the time, it opens the MQTT app instead. I was able to bring it up once, but have not been able to repeat. Also, fonts seem to be messed up on the display - non-half-height lines overlap with each other. My timezone info seems to be ignored (assumes I'm in Perth). Finally, the time did not seem to sync from NTP the one time I was able to run the NTP sync app.

Am I just bad at swiping? :P

wfdudley, 2020-11-04:
Eric,

Thanks for writing.  There haven't been a huge numbers of downloads
of my watch code, so I don't have data about how reliable it is.  Yours
is the first bug report.

There are debugging printfs in the routine that recognizes gestures, so you
can gather some debugging information.

If you edit appMQTT.cpp, line 164, and set DBG_GESTURES to 1, you'll
see messages printed to the Serial console in the Arduino IDE about the
gesture recognition.  To detect gestures, I record the max and min x and y
extents in a 100 msec interval (so, four numbers).  I then look for a
pattern
in the four numbers.  If the pattern is up, right, down, left, then the
gesture
is a CW circle (also matches starting at any point in that list and wrapping
around, so right, down, left, up is also CW circle).  Similarly, "up, left,
down,
right" is a CCW circle.

If the routine doesn't find a circle, then we look for a swipe in any of the
four cardinal directions.

Turn on DBG_GESTURES and see if that enlightens you, and let me know.

Bill Dudley
This email is free of malware because I run Linux.

lachesis, 2020-11-04:
here's the output of a two fast near-vertical swipes down with DEBUG_GESTURES 1
```
i = 100, p = 252
max_lrud[] = 110, 120, 1, 239
pts_lrud[] = 232, 1, 2, 233
x0 = 120, x = 110, xdir = -10
y0 = 1, y = 239, ydir = 238, amax = 238
maybe a circle
max_order[4] = 1, 2, 0, 3
dir_order = RULDRULD
found CCW CIRCLE
ccw circle

i = 100, p = 427
max_lrud[] = 99, 104, 5, 239
pts_lrud[] = 0, 407, 2, 408
x0 = 99, x = 104, xdir = 5
y0 = 5, y = 239, ydir = 234, amax = 234
maybe a circle
max_order[4] = 0, 2, 1, 3
dir_order = LURDLURD
found CW CIRCLE
cw circle
```

here's a slow vertical swipe down (which is not recognized at all?):
```
i = 100, p = 1356
max_lrud[] = 134, 149, 11, 239
pts_lrud[] = 0, 1336, 2, 1337
x0 = 134, x = 149, xdir = 15
y0 = 11, y = 239, ydir = 228, amax = 228
maybe a circle
max_order[4] = 0, 2, 2, 2
dir_order = LUUULUUU
```

here's what it looks like when I try to draw a CW circle (up, right, down, left):
```
i = 100, p = 1339
max_lrud[] = 37, 201, 69, 207
pts_lrud[] = 1321, 887, 471, 3
x0 = 75, x = 37, xdir = -38
y0 = 207, y = 195, ydir = -12, amax = 38
maybe a circle
max_order[4] = 3, 2, 1, 1
dir_order = DURRDURR
```

wfdudley, 2020-11-04:
Hi,

Thanks for that, I'll study it and get back to you.

Bill Dudley

This email is free of malware because I run Linux.

wfdudley, 2020-11-04:
I turned on DBG_GESTURE and tried your tests on my own watch.

For a swipe (up, down, left, or right, doesn't matter), I typically get
less than
100 points in 100 msec.  The first line in the output from your first test:

i = 100, p = 252

Says that the software captured 252 different x,y points in 100 msec.
This is very slow, compared to when I swipe.  I have to really work at
it to go slow enough to get more than 100 points in a swipe.
However, if I do that, then the software tries to make a circle out of
the gesture, which is obviously wrong.

To make the software less likely to mistake a slow swipe for a circle,
change
line 220 from:

if(points > 100) {  // is gesture a circle ?

to:

if(points > 100  && xmax > 60 && ymax > 60) {  // is gesture a circle ?

and that should make a difference.

Let me know.

Bill Dudley
This email is free of malware because I run Linux.

wfdudley, 2020-11-04:
correction to the "discussion" part:  The routine examines points
until the user stops touching the phone for 100 msec.  It does not
examine points for only 100 msec.  It will dutifully examine points for as
long as the user chooses to touch the screen.

Bill Dudley
This email is free of malware because I run Linux.

lachesis, 2020-11-04:
Excellent, that seems to have made it possible for me to make all 6 gestures. Thanks for the help. Now I can play with all the neat apps and maybe make my own Roku remote.

closed 2020-11-04 by lachesis

wfdudley, 2020-11-04:
The quick and dirty fix I posted last night broke swipes. I have uploaded an improved fix that seems to work well.