wfdudley/T-watch-2020 #10: Searching where to disable wake up on accel

sigmounte, 2020-11-21:
I have 8h of autonomy usng the default firmware , but i can clearly see that the wathc is waking up all the time , even when i don't need to look at time .  I'm totally of with having to use the push button to look at the time .

I just can't find where you programmed the waking up on accel ?

wfdudley, 2020-11-21:
I did not write that.  It's somewhere in the LVGL stuff, I don't know where.

Bill Dudley

sigmounte, 2020-11-21:
found it , was directly inside the .ino

thanks

wfdudley, 2020-11-21:
Yes, to be accurate, I didn't write it, I copied it from the demo watch code that the watch
ships with.  After sending my reply, I realized it was probably in that lump I copied into
the .ino file.

closed 2020-11-21 by wfdudley