wfdudley/T-watch-2020 #9: All Apps using wifi ( execpt NTP Time) work only once then do a DNS resolution error

sigmounte, 2020-11-21:
when using ntptime , everything work fine , and i can launch it multiple time : 

best strength = -36, best ssid = Sighq, tz = 506099284
best channel 11, best mac 1a:f6:cf:fd:97:70
铭德美家1-2301	mingde2301	1467951276
NETGEAR Qin32	Gutsmodernraven47	506099284
found SSID we know: Sighq
Connecting to WiFi network: Sighq
  password ######## channel 11 mac ##########
Waiting for WIFI connection...
waiting 2 seconds for wifi connection
WiFi connected! IP address: 192.168.43.239
WiFi connected! IP address: 192.168.43.239
Saturday, November 21 2020 12:25:58
printLocalTime(): UTC = 12:25:58 2020/11/21

when using any others app using wifi : i can do it only once ( ex bitcoin app ) : 

tzonestr = 506099284
best strength = -35, best ssid = Sighq, tz = 506099284
best channel 11, best mac 1a:f6:cf:fd:97:70
铭德美家1-2301	mingde2301	1467951276
NETGEAR Qin32	Gutsmodernraven47	506099284
found SSID we know: Sighq
Connecting to WiFi network: Sighq
  password ######## channel 11 mac #########
Waiting for WIFI connection...
WiFi connected! IP address: 192.168.43.239
WiFi connected! IP address: 192.168.43.239
WiFi connected! IP address: 192.168.43.239

the next time another time : 

Waiting for WIFI connection...
[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for api.coindesk.com
WiFi connected! IP address: 192.168.43.239
WiFi connected! IP address: 192.168.43.239

The only way to be able to start any wifi app without having this dns error is to start the ntptime before , look like it reset something that the others app don't ..

wfdudley, 2020-11-21:
No, this is incorrect. The Bitcoin app is the only one that has/causes difficulty.  The MQTT app and the new unreleased Weather app have no trouble connecting multiple times to wifi.

wfdudley, 2020-11-22:
I updated appBitcoin() to turn off the wifi when the app is exited.  This seems to fix this issue.

closed 2020-11-22 by wfdudley