wfdudley/T-watch-2020 #11: Having trouble compiling

XaosTX, 2020-11-30:
Getting the following errors when attempting to compile...

```
Arduino: 1.8.13 (Linux), Board: "TTGO T-Watch, Enabled, Default (2 x 6.5 MB app, 3.6 MB SPIFFS), 2000000, None"

sketch/SPIFF_fns.cpp: In function 'void build_acc_pts_file()':
SPIFF_fns.cpp:189:25: error: 'sizeof_Access_Points' was not declared in this scope
     for(int i = 0 ; i < sizeof_Access_Points ; i++) {
                         ^
sketch/SPIFF_fns.cpp: In function 'int read_acc_pts_file_and_compare_with_SSIDs(WiFiAp*)':
SPIFF_fns.cpp:230:27: error: 'number_of_networks' was not declared in this scope
       for (int i = 0; i < number_of_networks; ++i) {
                           ^
Multiple libraries were found for "WiFi.h"
 Used: /home/XXXX/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /opt/arduino-1.8.13/libraries/WiFi
Multiple libraries were found for "SD.h"
 Used: /home/XXXX/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SD
 Not used: /opt/arduino-1.8.13/libraries/SD
exit status 1
'sizeof_Access_Points' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

wfdudley, 2020-11-30:
fixed.  I have to keep two versions of the file, one for me,
one for the rest of the world.  I didn't keep the second
one updated.

closed 2020-11-30 by wfdudley

nate1232, 2020-12-01:
Thanks!  I had the same issue when starting to play with this.