wfdudley/T-watch-2020 #15: AceTime compiling error

Lacybad, 2020-12-27:
Hello!

I have a little problem. I am trying to compile the software in Arduino IDE. I have all the neccesary up to date libraries, but the IDE is giving compiling errors. I am using the IDE version 1.8.13 with the latest ESP32 definitions on Win10. I could not solve this issue on my own. Any help or suggestion would be really appreciated. Thank you in advance. Lazlo

AceTime 1.3.0 
custom_lib.a(zone_registry.cpp.o):(.rodata._ZN8ace_time6zonedb13kZoneRegistryE+0x42c): undefined reference to `ace_time::zonedb::kZoneWET'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling to TTGO T-Watch .

Or another error with the AceTime-develop library

custom_lib.a(my_tz.cpp.o):(.literal.startup._GLOBAL__sub_I_tz_opts+0x4): undefined reference to `ace_time::zonedb::kZoneRegistry'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling to TTGO T-Watch .

wfdudley, 2020-12-27:
I built DudleyWatch using AceTime 1.1.0.  I just now upgraded my AceTime library to 1.3.0, and it compiles without errors.  That suggests that your installation of AceTime is damaged in some way.  AceTime 1.3.0 does require a new additional library, AceCommon, so make sure you have that installed as well.

closed 2020-12-27 by wfdudley

Lacybad, 2021-01-03:
Hi!

I have found what was the problem  :) Not a library issue, was something with Win10, reinstalled IDE again and now it iz working fine ^^ .