wfdudley/T-watch-2020 #16: Compilation Error: Sketch uses 1839173 bytes (140%) of program storage space.

nickw56, 2020-12-30:
When I compile I get the following error: 

"Arduino: 1.8.13 (Windows 10), Board: "TTGO T1, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

Sketch uses 1839173 bytes (140%) of program storage space. Maximum is 1310720 bytes.text section exceeds available space in board

Global variables use 116472 bytes (35%) of dynamic memory, leaving 211208 bytes for local variables. Maximum is 327680 bytes.

Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.

Error compiling for board TTGO T1."

Has anyone else had the same problem ? and how did you fix it.  I would be happy to lose some of the extensive range of features but I don't know where to start doing that.

genotix, 2020-12-30:
Looks like you have way to much Global variables set.
Make sure to define more static and const types.
Make sure to omit the use of String types.
Whenever you use statements like Serial prints that show values, try to do it like this:
Serial.print(F(“Temperature value:”)); // The F() function places the static value in STATIC MEMORY
Serial.println(integerValueTemperature); // This “embeds” the String and Integer value in a single message
Serial.flush(); // This flushes the information before continuing (just a best practise)

Try to reuse larger variables and do what is advised; check the Sketch too big URL.

Regards,

Eric

wfdudley, 2020-12-31:
Your partitioning is wrong.  You need to use the special partitioning for the T-watch-2020, which is:
2x6.5 MB APP, 1.5MB SPIFFS
I am fully aware of all the things genotix has mentioned, but thank you for the suggestions.

closed 2020-12-31 by wfdudley

nickw56, 2020-12-31:
Thanks for your reply.  Where/how do I change this ?  In my Arduino IDE/tools/partioning scheme there is no option that matches this and I can't see an option to add a new scheme. 

Apologies if I am missing something obvious - I am a 64 year old novice so your patience and assistance is much appreciated.

wfdudley, 2020-12-31:
I just did a completely fresh install of Arduino and "all the fixin's" to
see how this worked.

Once I installed the esp32 support package, and selected "TTGO T-watch"
from the "espArduino" in the "Tools/Board:" dropdown menu,
the proper partition scheme was automatically set.  I didn't have to touch
this option.

Bill Dudley

This email is free of malware because I run Linux.

nickw56, 2020-12-31:
Hi Bill

I got it working using one of the Huge App scheme.  It's absolutely the best T-watch version I have tried so far.  If I encounter any further issues I will try re-installing the Arduino IDE and see if that gives me the scheme you reccommended.

In the meantime- many thanks and all the best for 2021

Nick