wfdudley/T-watch-2020 #18: Got my first issue

Therealjosephchrzempiec, 2021-01-02:
Hello i have i have tried to follow everything you wrote to the letter. however i get this error.

`In file included from C:\Users\jJcSystems\Documents\Arduino\libraries\AceTime-develop\src/AceTime.h:20:0,

                 from C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020-master\DudleyWatch\DudleyWatch.ino:26:

C:\Users\jJcSystems\Documents\Arduino\libraries\AceTime-develop\src/ace_time/common/compat.h:78:25: fatal error: AceCommon.h: No such file or directory

compilation terminated.

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\jJcSystems\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Users\jJcSystems\Desktop\Bakup\12-24-2020\Desktop\arduino-1.8.3\libraries\WiFi
exit status 1
Error compiling for board TTGO T-Watch.

Joseph`

Therealjosephchrzempiec, 2021-01-02:
I do not se this file at all. The AceCommon.h

Joseph

wfdudley, 2021-01-02:
The AceCommon library is required by the AceTime library.  This is stated in my README.md.  When I installed AceTime, the library manager automatically asked for permission to install AceCommon as well.  So install it.

closed 2021-01-02 by wfdudley

Therealjosephchrzempiec, 2021-01-02:
Hello i didn't see the last part where the common part was missing in yor readme. But now i do have a new problem. 

```
WARNING: Category 'other' in library OpenWeatherOneCall is not valid. Setting to 'Uncategorized'
In file included from C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020-master\DudleyWatch\DudleyWatch.ino:33:0:

C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020-master/DudleyWatch.h:44:8: error: 'RTC_Date' does not name a type

 EXTERN RTC_Date tnow;

        ^

C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020-master/DudleyWatch.h:129:8: error: 'TFT_eSPI' does not name a type

 EXTERN TFT_eSPI *tft;

        ^

C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020-master/DudleyWatch.h:130:8: error: 'AXP20X_Class' does not name a type

 EXTERN AXP20X_Class *power;

        ^

C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020-master/DudleyWatch.h:230:8: error: 'TTGOClass' does not name a type

 EXTERN TTGOClass *ttgo;

        ^

C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020-master\DudleyWatch\DudleyWatch.ino: In function 'void update_step_counter()':

DudleyWatch:47: error: 'ttgo' was not declared in this scope

   step_counter = ttgo->bma->getCounter();

                  ^

C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020-master\DudleyWatch\DudleyWatch.ino: In function 'void resetStepCounter()':

DudleyWatch:51: error: 'ttgo' was not declared in this scope

   ttgo->bma->resetStepCounter();

   ^

C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020-master\DudleyWatch\DudleyWatch.ino: In function 'void bright_check()':

DudleyWatch:57: error: 'power' was not declared in this scope

   if (power->isVBUSPlug()) {

       ^

DudleyWatch:66: error: 'ttgo' was not declared in this scope

     ttgo->setBrightness(screen_brightness);       // 0-255

     ^

C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020-master\DudleyWatch\DudleyWatch.ino: In function 'void low_energy()':

DudleyWatch:78: error: 'ttgo' was not declared in this scope

     if (ttgo->bl->isOn()) {

         ^

DudleyWatch:95: error: 'AXP202_INT' was not declared in this scope

[... 143 lines cut ...]
DudleyWatch:434: error: 'power' was not declared in this scope
power->readIRQ();
^
DudleyWatch:539: error: 'ttgo' was not declared in this scope
 ttgo->tft->fillScreen(TFT_BLACK);
   ^
DudleyWatch:539: error: 'TFT_BLACK' was not declared in this scope
  ttgo->tft->fillScreen(TFT_BLACK);
DudleyWatch:540: error: 'tft' was not declared in this scope
   tft->setTextSize(1);
   ^
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\jJcSystems\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Users\jJcSystems\Desktop\Bakup\12-24-2020\Desktop\arduino-1.8.3\libraries\WiFi
Multiple libraries were found for "SD.h"
 Used: C:\Users\jJcSystems\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD
 Not used: C:\Users\jJcSystems\Desktop\Bakup\12-24-2020\Desktop\arduino-1.8.3\libraries\SD
exit status 1
'ttgo' was not declared in this scope

```

I only have one Esp wifi library.  The rest I'm unsure of because it is looking for SD not being used.

Joseph

wfdudley, 2021-01-03:
I have no idea how you're getting all these errors.  Perhaps computer programming isn't the best hobby for you?

Therealjosephchrzempiec, 2021-01-03:
That is what I'm goibg to try again. I think some is conflitctig with my other libraries.

Joseph

Therealjosephchrzempiec, 2021-01-03:
Hello when i get to the NTPClient library. I'm having trouble with fining the library from the link. Can you help me with that please?

Joseph

Therealjosephchrzempiec, 2021-01-03:
Hello i deleted all libraries and sketches. I get the same Errors when i downloaded all libraries included yours. I try to compile and i get same errors.

In file included from C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020-master\DudleyWatch\DudleyWatch.ino:33:0:
C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020-master/DudleyWatch.h:44:8: error: 'RTC_Date' does not name a type
 EXTERN RTC_Date tnow;
        ^
C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020-master/DudleyWatch.h:129:8: error: 'TFT_eSPI' does not name a type
 EXTERN TFT_eSPI *tft;

And so forth.

Joseph

Fra4prg, 2021-01-03:
Hello Joseph,

I would recommend to uninstall the Arduino IDE including the deletion of the sketch folder (probably C:\Users\jJcSystems\Documents\Arduino but verify the path in your preferences) 
as well as the Appdata folder, probably C:\Users\jJcSystems\AppData\Local\Arduino15 in your case.

After this install  Arduino IDE 1.8.13 again.
Then  add the following path 
https://dl.espressif.com/dl/package_esp32_index.json 
under file -> preferences -> additional URLs, 
go to Tools->Board->Boards Manager and search for Esp32 -> install Esp32 (Espressif Systems) V1.0.4
Download the TTGO ZIP library from 
https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library
and install it via Sketch->include  library ->ZIP library

Follow the instructions of wfdudley to install the required libraries.
The NTPCliesnt library can also be found in Github at 
https://github.com/arduino-libraries/NTPClient

Now everything should work well.
This procedure also helped in my case where I got some strange compiler errors in the Windows IDE which did not show under Linux.

Regards
Frank

Therealjosephchrzempiec, 2021-01-04:
I already uninstalled IDE in windows 10 and reinstalled it. That made no difference this is strange. Firt thing i thought of was he IDE i was running 1.8.13 the latest. I normally install the latest. I don't do a upgrade of IDE. 

Joseph

Therealjosephchrzempiec, 2021-01-04:
I will try it again. Thnk you

Joseph

Therealjosephchrzempiec, 2021-01-04:
Hello the crazy thing is i can load TTGO watch library and the example codes with no problem. It's somehow i guess the code that comes from yours I'm having problems with. I know it is something I'm doing wrong. I'm working on it. 

Joseph