wfdudley/T-watch-2020 #21: WiFi.onEvent(WiFiEvent); should only be called once

FedericoBusero, 2021-01-04:
Each time the app 'NTP Time' or 'Weather' is executed, `WiFi.onEvent(WiFiEvent); ` is executed, but this should only be called once at setup time. If the app NTP time has been opended 10 times, a wifi connection that is created will result in 10x calling the function WiFiEvent, which can be seen on the debug window: you will see 10 times

WiFi connected! IP address: ...

wfdudley, 2021-01-04:
Can you explain more about this problem?  What is wrong with calling WiFi.onEvent() multiple times?  Does it constitute a memory leak?

FedericoBusero, 2021-01-04:
The memory allocation will grow each time you call onEvent: it will add a function to the onEvent-callback-list. If you have run the NTP-app 100 times, you will have 100 functions in the onEvent callback list. This means that if you get a connection, the function "WiFiEvent" will be called 100 times!

You can see this in the debug window, you will see the following list growing
WiFi connected! IP address: ...
WiFi connected! IP address: ...
WiFi connected! IP address: .....
WiFi connected! IP address: .....
WiFi connected! IP address: .....

Solution
- only call WiFi.onEvent once
OR
- call Wifi.removeEvent when finished

`WiFiEventId_t eventID = WiFi.onEvent(WiFiEvent);`
...
`WiFi.removeEvent(eventID);`

wfdudley, 2021-01-04:
Thanks for the explanation.  I like the "removeEvent()" solution.  I'll
work on that.

And thanks again for all your help.

Bill Dudley
This email is free of malware because I run Linux.

wfdudley, 2021-01-05:
fixed with today's commit.  I used WiFi.removeEvent().

closed 2021-01-05 by wfdudley