wfdudley/T-watch-2020 #25: Uploading error.

Therealjosephchrzempiec, 2021-01-06:
Hello i have all the same libraries as in your list and that from other sites. I can upload Dan code that this was based off of. however i get a error when trying to upload. Error is below. I never saw any of theses errors before.

`Invalid library found in C:\Users\jJcSystems\Documents\Arduino\libraries\TTGO-T-Watch-master: no headers files (.h) found in C:\Users\jJcSystems\Documents\Arduino\libraries\TTGO-T-Watch-master
In file included from C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020\DudleyWatch\DudleyWatch.ino:33:0:
C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020/DudleyWatch.h:44:8: error: 'RTC_Date' does not name a type
 EXTERN RTC_Date tnow;
        ^
C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020/DudleyWatch.h:130:8: error: 'TFT_eSPI' does not name a type
 EXTERN TFT_eSPI *tft;
        ^
C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020/DudleyWatch.h:131:8: error: 'AXP20X_Class' does not name a type
 EXTERN AXP20X_Class *power;
        ^
C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020/DudleyWatch.h:231:8: error: 'TTGOClass' does not name a type
 EXTERN TTGOClass *ttgo;
        ^
C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020\DudleyWatch\DudleyWatch.ino: In function 'void update_step_counter()':
DudleyWatch:47:18: error: 'ttgo' was not declared in this scope
   step_counter = ttgo->bma->getCounter();
                  ^
C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020\DudleyWatch\DudleyWatch.ino: In function 'void resetStepCounter()':
DudleyWatch:51:3: error: 'ttgo' was not declared in this scope
   ttgo->bma->resetStepCounter();
   ^
C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020\DudleyWatch\DudleyWatch.ino: In function 'void bright_check()':
DudleyWatch:57:7: error: 'power' was not declared in this scope
   if (power->isVBUSPlug()) {
       ^
DudleyWatch:66:5: error: 'ttgo' was not declared in this scope
     ttgo->setBrightness(screen_brightness);       // 0-255
     ^
C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020\DudleyWatch\DudleyWatch.ino: In function 'void low_energy()':
DudleyWatch:78:9: error: 'ttgo' was not declared in this scope
     if (ttgo->bl->isOn()) {
         ^
DudleyWatch:95:53: error: 'AXP202_INT' was not declared in this scope
             erret = gpio_wakeup_enable ((gpio_num_t)AXP202_INT, GPIO_INTR_LOW_LEVEL);
                                                     ^
DudleyWatch:99:53: error: 'BMA423_INT1' was not declared in this scope
             erret = gpio_wakeup_enable ((gpio_num_t)BMA423_INT1, GPIO_INTR_HIGH_LEVEL);
                                                     ^
DudleyWatch:135:35: error: 'lv_disp_trig_activity' was not declared in this scope
         lv_disp_trig_activity(NULL);
                                   ^
C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020\DudleyWatch\DudleyWatch.ino: In function 'void Serial_timestamp()':
DudleyWatch:179:29: error: 'ttgo' was not declared in this scope
   Serial.printf("%s UTC\n", ttgo->rtc->formatDateTime());
                             ^
C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020\DudleyWatch\DudleyWatch.ino: In function 'void setup()':
DudleyWatch:213:16: error: 'EEPROM_SIZE' was not declared in this scope
   EEPROM.begin(EEPROM_SIZE);
                ^
DudleyWatch:253:3: error: 'ttgo' was not declared in this scope
   ttgo = TTGOClass::getWatch();
   ^
DudleyWatch:253:10: error: 'TTGOClass' has not been declared
   ttgo = TTGOClass::getWatch();
          ^
DudleyWatch:255:3: error: 'tft' was not declared in this scope
   tft = ttgo->tft; // just a shorthand for ttgo->tft
   ^
DudleyWatch:256:3: error: 'power' was not declared in this scope
   power = ttgo->power; // just a shorthand for ttgo->power
   ^
DudleyWatch:258:19: error: 'TFT_BLACK' was not declared in this scope
   tft->fillScreen(TFT_BLACK);
                   ^
DudleyWatch:259:21: error: 'TFT_YELLOW' was not declared in this scope
   tft->setTextColor(TFT_YELLOW, TFT_BLACK); // Note: the new fonts do not draw the background colour
                     ^
DudleyWatch:264:21: error: 'AXP202_BATT_VOL_ADC1' was not declared in this scope
   power->adc1Enable(AXP202_BATT_VOL_ADC1 | AXP202_BATT_CUR_ADC1 | AXP202_VBUS_VOL_ADC1 | AXP202_VBUS_CUR_ADC1, AXP202_ON);
                     ^
DudleyWatch:264:44: error: 'AXP202_BATT_CUR_ADC1' was not declared in this scope
   power->adc1Enable(AXP202_BATT_VOL_ADC1 | AXP202_BATT_CUR_ADC1 | AXP202_VBUS_VOL_ADC1 | AXP202_VBUS_CUR_ADC1, AXP202_ON);
                                            ^
DudleyWatch:264:67: error: 'AXP202_VBUS_VOL_ADC1' was not declared in this scope
   power->adc1Enable(AXP202_BATT_VOL_ADC1 | AXP202_BATT_CUR_ADC1 | AXP202_VBUS_VOL_ADC1 | AXP202_VBUS_CUR_ADC1, AXP202_ON);
                                                                   ^
DudleyWatch:264:90: error: 'AXP202_VBUS_CUR_ADC1' was not declared in this scope
   power->adc1Enable(AXP202_BATT_VOL_ADC1 | AXP202_BATT_CUR_ADC1 | AXP202_VBUS_VOL_ADC1 | AXP202_VBUS_CUR_ADC1, AXP202_ON);
                                                                                          ^
DudleyWatch:264:112: error: 'AXP202_ON' was not declared in this scope
   power->adc1Enable(AXP202_BATT_VOL_ADC1 | AXP202_BATT_CUR_ADC1 | AXP202_VBUS_VOL_ADC1 | AXP202_VBUS_CUR_ADC1, AXP202_ON);
                                                                                                                ^
DudleyWatch:265:20: error: 'AXP202_VBUS_REMOVED_IRQ' was not declared in this scope
   power->enableIRQ(AXP202_VBUS_REMOVED_IRQ | AXP202_VBUS_CONNECT_IRQ | AXP202_CHARGING_FINISHED_IRQ, AXP202_ON);
                    ^
DudleyWatch:265:46: error: 'AXP202_VBUS_CONNECT_IRQ' was not declared in this scope
   power->enableIRQ(AXP202_VBUS_REMOVED_IRQ | AXP202_VBUS_CONNECT_IRQ | AXP202_CHARGING_FINISHED_IRQ, AXP202_ON);
                                              ^
DudleyWatch:265:72: error: 'AXP202_CHARGING_FINISHED_IRQ' was not declared in this scope
   power->enableIRQ(AXP202_VBUS_REMOVED_IRQ | AXP202_VBUS_CONNECT_IRQ | AXP202_CHARGING_FINISHED_IRQ, AXP202_ON);
                                                                        ^
DudleyWatch:269:25: error: 'AXP202_EXTEN' was not declared in this scope
   power->setPowerOutPut(AXP202_EXTEN, AXP202_OFF);
                         ^
DudleyWatch:269:39: error: 'AXP202_OFF' was not declared in this scope
   power->setPowerOutPut(AXP202_EXTEN, AXP202_OFF);
                                       ^
DudleyWatch:270:25: error: 'AXP202_DCDC2' was not declared in this scope
   power->setPowerOutPut(AXP202_DCDC2, AXP202_OFF);
                         ^
DudleyWatch:271:25: error: 'AXP202_LDO3' was not declared in this scope
   power->setPowerOutPut(AXP202_LDO3, AXP202_OFF); // audio device
                         ^
DudleyWatch:272:25: error: 'AXP202_LDO4' was not declared in this scope
   power->setPowerOutPut(AXP202_LDO4, AXP202_OFF);
                         ^
DudleyWatch:283:11: error: 'BMA423_INT1' was not declared in this scope
   pinMode(BMA423_INT1, INPUT);
           ^
DudleyWatch:304:11: error: 'AXP202_INT' was not declared in this scope
   pinMode(AXP202_INT, INPUT);
           ^
C:\Users\jJcSystems\Documents\Arduino\libraries\T-watch-2020\DudleyWatch\DudleyWatch.ino: In function 'void loop()':
DudleyWatch:392:12: error: 'ttgo' was not declared in this scope
    rlst =  ttgo->bma->readInterrupt();
            ^
DudleyWatch:397:2: error: 'power' was not declared in this scope
  power->readIRQ();
  ^
DudleyWatch:425:12: error: 'ttgo' was not declared in this scope
    rlst =  ttgo->bma->readInterrupt();
            ^
DudleyWatch:428:8: error: 'ttgo' was not declared in this scope
    if (ttgo->bma->isStepCounter()) {
        ^
DudleyWatch:434:2: error: 'power' was not declared in this scope
  power->readIRQ();
  ^
DudleyWatch:539:7: error: 'ttgo' was not declared in this scope
       ttgo->tft->fillScreen(TFT_BLACK);
       ^
DudleyWatch:539:29: error: 'TFT_BLACK' was not declared in this scope
       ttgo->tft->fillScreen(TFT_BLACK);
                             ^
DudleyWatch:540:7: error: 'tft' was not declared in this scope
       tft->setTextSize(1);
       ^
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\jJcSystems\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
Multiple libraries were found for "SD.h"
 Used: C:\Users\jJcSystems\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
Multiple libraries were found for "Ethernet.h"
 Used: C:\Users\jJcSystems\Documents\Arduino\libraries\Ethernet
 Not used: C:\Program Files (x86)\Arduino\libraries\Ethernet
exit status 1
'ttgo' was not declared in this scope
`

Joseph

wfdudley, 2021-01-06:
I don't know how you and the other guy have this problem, but it's not the code.

Something is horribly defective with your installation.

Read the beginner docs here:

https://github.com/wfdudley/T-watch-2020/blob/master/docs/how-to-build-watch.md

closed 2021-01-06 by wfdudley

Therealjosephchrzempiec, 2021-01-06:
I can upload the lillygo watch library and code i can upload the example default code with no problem. I can also upload Dan Geiger code with no problem.

Joseph

Therealjosephchrzempiec, 2021-01-06:
Let me try again.

wfdudley, 2021-01-06:
If you figure it out, let me know.

There's another guy who gets the same million errors as you.

Bill Dudley

This email is free of malware because I run Linux.