wfdudley/T-watch-2020 #26: get_location.cpp doesn't compile

FedericoBusero, 2021-01-07:
```
sketch\get_location.cpp: In function 'int get_city_from_lat_long(char*, char*)':
get_location.cpp:168:94: error: 'LOCATIONIQ_KEY' was not declared in this scope
   sprintf(url, "https://us1.locationiq.com/v1/reverse.php?key=%s&format=json&lat=%s&lon=%s", LOCATIONIQ_KEY, latitude, longitude);
```
...

`'LOCATIONIQ_KEY' was not declared in this scope`

wfdudley, 2021-01-07:
add this line to personal_info.h:

#define LOCATIONIQ_KEY ""

I can't make the mod now, busy with real work.

Thanks,
Bill Dudley

This email is free of malware because I run Linux.

wfdudley, 2021-01-07:
I made the change to personal_info.h

dud

This email is free of malware because I run Linux.

closed 2021-01-09 by FedericoBusero