wfdudley/T-watch-2020 #27: unable to push a new branch

Fra4prg, 2021-01-08:
[modfiles.zip](https://github.com/wfdudley/T-watch-2020/files/5788652/modfiles.zip)
Hello Bill,

first of all thank you for this great project.
It’s the best T-Watch I found so far.

I made some additions to the stop watch and I would appreciate if you could merge this into your project.
Please apologise, I'm new to Github and it was a nightmare to configure everything but I still get an error when I try to push my branch: "remote:Permission to wfdudley/T-watch-2020.git denied to Fra4prg..."
I don't know: Isn't it allowed to submit a branch to a foreign project?
How else could I submit my proposal?
I added my changed/added files here as modfiles.zip. All changes are marked with the comment // Fra4prg:

Description of my changes:
Added "Smart Sound" option ("HeliTimer")
"Smart Sound" enables you to hear the number of minutes without having to look on the display.
The idea behind this is that human brain can identify the number of 3 beeps as triple without counting them even if they are very short and fast.
So up to 3 beeps are played slowly and preceeding triples are faster.
For every 10 minutes a sweep sound is preceeding.
A new top left button switches modes between [no sound] - [vibration + sound] - [only sound]

Some examples to show the sequence:
Minute  1: C5
Minute  2: C5-E5
Minute  3: C5-E5-G5
Minute  4: triple - C5
Minute  5: triple - C5-E5
...
Minute  9: triple - triple - C5-E5-G5
Minute 10: sweep
...
Minute 18: sweep - triple - triple - C5-E5
and so on...

As you can see this works well for short times up to 10 minutes then it gets weird...
The idea came up when I was flying model helicopters where I had to control the flight time due to small batteries and must not loose visual contact to my heli model.
So I made a small Arduino based timer called HeliTimer.
But this function fits perfect to the T-watch :-)

wfdudley, 2021-01-08:
Mr no name,
(I didn't see anything that looks like a name in your email).

I'm not a github expert, I only started using it in earnest with this
project.
But usually, when people want to submit code to a project, they generate
a "pull request".  Then I get an alert of the pull request, and review the
changes, and then decide if I'll accept the code.

So I'd suggest researching how to do a pull request.  I did one with the
Sharango
T-watch code, and it wasn't that hard to figure out.

I'll leave your issue open for now.

Bill Dudley

This email is free of malware because I run Linux.

Fra4prg, 2021-01-09:
Oops!
My name is Frank Scholl.
Sorry that I missed to put my name here - very embarrassing  8-O
Yes I tried a pull request but as I understood this requires a branch.
Maybe creating a fork is the right way?
I will investigate more on the  pull request.
Thank you for your patience.

Best regards
Frank

wfdudley, 2021-01-09:
Frank,

Yes, you create your own branch of the code in your github account, make
your changes, and then do a pull request.

Bill

This email is free of malware because I run Linux.

Fra4prg, 2021-01-09:
Hi Bill,

I guess I managed to create my pull request via fork.
If everything looks ok in this process you can close this issue.

Regards 
Frank

closed 2021-01-11 by Fra4prg

wfdudley, 2021-01-11:
Thanks.  I'll get to your pull request soon, but I was busy with another
problem all weekend.

Bill Dudley

This email is free of malware because I run Linux.