wfdudley/T-watch-2020 #29: Add License

guyou, 2021-01-30:
This is a nice project.
But a LICENSE file is missing, helping people to reuse it while following your own choices.

wfdudley, 2021-01-30:
I have avoided that because there are many choices and I don't want to
spend the time
researching it.  Can you recommend a site that gives a good overview of
open source
licenses?  licenses-for-dummies.com?

Bill Dudley

AffekeNommu, 2021-01-31:
GitHub has that covered.
[https://docs.github.com/en/github/creating-cloning-and-archiving-repositories/licensing-a-repository](https://docs.github.com/en/github/creating-cloning-and-archiving-repositories/licensing-a-repository)

wfdudley, 2021-01-31:
Thanks all.  I've chosen BSD 3-Clause.

closed 2021-01-31 by wfdudley

guyou, 2021-01-31:
I discovered a site like your hypothetical licences-for-dummies: https://choosealicense.com/
This is one of major problem with open-source: too many choices. ;-)