wfdudley/T-watch-2020 #33: Unable to connect to Wifi

RChadwick7, 2021-02-04:
I modified the my_WiFi.h to look like this:

EXTERN WIFIAP AccessPoints[]
#ifdef __WIFIMAIN__
= {
    { "Guest123", "ThisIsmypassword1", TZ_EST },		// Home
    { "Not Your A10", "thisismypassword2", TZ_EST },	// my phone hotspot
    { "work-ssid", "password4job", TZ_EST },	// my job's WiFi
    { "friend #1 ssid", "pass#2word", TZ_PST },	// friend one's WiFi
    { "2nd friend SSID", "passwo#4rd", TZ_CST }, // friend two's WiFi
    { "dentist-ssid", "pa#5ssword", TZ_EST },	// my dentist's WiFi
    { "\xE9\x93\xAD\xE5\xBE\xB7\xE7\xBE\x8E\xE5\xAE\xB6" "1-2301", "mingde2301", TZ_CHINA },    // an AirBnB in China, kept as example of non-ASCII chars
    { "friend 3 SSID", "passwor#6d", TZ_EST }	// friend three's WiFi
}
#endif

Unfortunately, It never connects to wifi. If I tap on 'weather', it finds all nearby access points, including the ones I configured, but it then freezes at this screen. If I tap on 'delete wifi', I see the pre-configured AP's, but not mine. I then changed the MQTT default password to see if it changed, and it didn't. After making the changes, I saved the file, recompiled, and then uploaded. I'm guessing I'm doing something wrong, or need to somehow wipe the watch before uploading?

RChadwick7, 2021-02-04:
I removed and replaced the battery, and it's working now.

closed 2021-02-04 by RChadwick7

wfdudley, 2021-02-05:
How do you remove the battery?

You can wipe the stored list of access points by uncommenting this line:

// deleteFile(SPIFFS, "/acc_pts.txt");

in function build_acc_pts_file() in file SPIFF_fns.cpp.
uncomment it, compile and upload, and then do any wifi-thing.
The file in SPIFFS will be re-created from the AccessPoints data.
Make sure you re-comment it and re-compile and re-upload
afterwards, or you'll never be able to store a new WiFi access
point in the watch.

Bill Dudley

This email is free of malware because I run Linux.

RChadwick7, 2021-02-05:
Thanks for the response, and a very interesting project. As for the battery, at least on mine, I remove the black back, then lift the battery out from one side. I actually found and bought a second battery on Ebay.