wfdudley/T-watch-2020 #34: Err for dayOfWeek with Lilygo Skin

surshis, 2021-02-10:
Thanks for the hard work, the code works fantastic, watch looks good. Only issue I am having is for the Day of week in the Lilygo skin, it shows Err everyday. Cannot get the day of week showing correctly. Date, month, year shows correctly. Cannot find the issue, thanks.

wfdudley, 2021-02-10:
Fixed.  When I change the code on my development machine, it helps if I upload the changes to github.
Anyway, you bug has told me that pretty much nobody is using the LilyGo watch face.

closed 2021-02-10 by wfdudley

RChadwick7, 2021-02-10:
I use it, also noticed the err, but didn't know it was for the day of the week :)

surshis, 2021-02-11:
Thanks a lot for the fix. Working now. Lilygo watch face is the primary reason I bought the watch and installed your code. Although I did customize it. Thanks

wfdudley, 2021-02-11:
What customizations have you done?  Are they worth putting back in the code?

Bill Dudley
This email is free of malware because I run Linux.

surshis, 2021-02-12:
Its not much. Just changed some stuff on the lilygo watch face only. Moved text, changed background to black  and removed stuff to have less crowded look. Nothing technical. Still learning the programming. I do have a question if you don't mind. I turned off wakeup on motion to save battery but the tap to wakeup has stopped working. Can I change the sensitivity of Wakeup on motion somewhere to only wakeup with a hard shake? Thanks

wfdudley, 2021-02-12:
I don't know if the sensitivity of wake on motion can be altered.  The
source code for the "drivers" for
the peripherals (touch, motion) is included when you download ESP32
support, so you can poke around
in there and figure it out.  That's how I figured out that gestures are
handled in user code; the touch
sensor says it recognizes gestures, but the chip maker never bothered to
implement the feature.

You can always make your "new" watch face a fifth choice in the skin select.

Bill Dudley
This email is free of malware because I run Linux.