wfdudley/T-watch-2020 #35: Cannot find the implementation of draw_keyboard

Build1971, 2021-03-06:
Dear William,

I got your code working on my TTGO watch.
It is really great fun and I am making changes to your code so that the watch looks and works the way I want it. 

So far, I have fixed the temperature meter (was in the far left of the screen and is now on the coloured scale).
I also increased the font size of the battery stats.

Now I would like to increase the font size of the menu buttons, but I cannot find the place where these buttons are drawn on the screen. I did find the spot where I can change the labels (which will be needed if I finally succeed to increase the font).
I also found the spot where the calculator buttons are drawn on the screen (there the font size is 4, which is OK).
So my question to you is: where can I find the implementation of the blue button menu?

I found this in the header file, but cannot find the implementation of this function.

void draw_keyboard(uint8_t, const char **, uint8_t, bool, char *);

I found out that the third argument is the font size. But what does the bool do? And could I somehow add pictograms instead of text?

Hope you can help me out, since I do not know where to look further.

Kind regards,

Koen.

wfdudley, 2021-03-07:
Koen,

draw_keyboard() is in appSetTime.cpp, because it's derived from Dan
Geiger's time setting routine's
keyboard.  I enhanced it to make it more flexible.  I should re-organized
the code, but that's not "fun",
so I haven't.

One of the arguments to draw_keyboard() is "font", so you do have some
control over the font used
on the keyboard already.

You might find it useful to acquire and learn use of the "grep" program for
searching text.  Not remembering
precisely where draw_keyboard was defined, I searched for it by doing:

grep draw_keyboard *.cpp

A more precise search would have been:

grep "void draw_keyboard" *.cpp

And if you really had no idea where it was, then in Unix/Linux, you could
do this:

find . -type f -print0 | xargs -0 grep "void draw_keyboard"

And that would find it even if the source was in a subdirectory.  (That
requires the find, xargs, and grep
programs.  If you run Winders, then you can still do this by installing
cygwin (https://www.cygwin.com)).

Bill Dudley

This email is free of malware because I run Linux.

Build1971, 2021-03-07:
Hi Bill,

Thanks a lot. That was one of the few places where I did not look…

Thanks as well for the find . -type f -print0 | xargs -0 grep “void draw_keyboard” hint. It works on MacOS as well. This will be very useful the next time.

… some restructuring of the TT-watch project would indeed be a good thing to do. But I can imagine you are not looking forward to that task. | did not dare to start it, because to me all the #define’s are a bit abacadabra. So I am afraid that after removing a lot of stuff I will end op with a sketch that does not compile and will give a very vague hint on what might be wrong. Of course I could remove things one block at  a time, but compilation of the current sketch takes a few minutes. So that would take a lot of time. And… a sketch that is compiled without errors does not necessarily do what it is supposed to do...
So, I will focus on the things I want to change first. Also because it will help me to understand how everything was put together. Anyway, now I have the place where I can change the menu!

Kind regards,

Koen.

On 7 Mar 2021, at 05:05, William F. Dudley Jr. <notifications@github.com<mailto:notifications@github.com>> wrote:

find . -type f -print0 | xargs -0 grep "void draw_keyboard"

wfdudley, 2021-03-08:
There are really two sets of UI code -- the "official" stuff, which I use
in the "settings" dialogs, and the
"light weight" stuff, draw_keyboard() etc, that is derived from Dan
Geiger's work.  That "light weight"
UI code could stand to be cleaned up, better integrated and organized,
etc,, but it's not fun and I'm
not getting paid for this, so why bother?  By "organized" I mean putting
all the keyboard and menu stuff
in it's own module.

The "official" UI code is very capable, but slow.  The "official" watch
code is a real pig, IMHO.  It's
very slow to respond to user input.

Bill Dudley
This email is free of malware because I run Linux.

wfdudley, 2021-03-12:
answered.

closed 2021-03-12 by wfdudley