wfdudley/T-watch-2020 #38: my_tz.cpp:11:8: error: 'ace_time::BasicZoneManager' is not a template

yonggithub, 2022-04-01:
My device is T-watch-2020 v2.
Unable to compile this project, and prompt an error: `my_tz.cpp:11:8: error: 'ace_time::BasicZoneManager' is not a template`.

Do any process I miss to do?  

[image]

below is my config.h  
[image]

wfdudley, 2022-04-01:
Hi.

I don't have that watch, and, more importantly, I didn't write the Ace-time
library.

Good luck.
Bill Dudley

sent from my Linux phone

czesioG, 2022-04-18:
Hi,
simply change acetime library to 1.8.2. For all above the code needs to be rewritten  - https://github.com/bxparks/AceTime/blob/develop/MIGRATING.md#MigratingToVersion190
BTW I use 2020 v3 and the code works (at least it was compiled ok and uploded)